ferrous-systems/rust-exercises #84: Note: use of static mut is a warning in 1.77

jonathanpallant, 2024-03-08:
Need to make static mut go away, or disable deny(warnings)

tallamjr, 2024-03-08:
Ref:
```bash
15:02:55 ✔ ~/github/tallamjr/forks/rust-exercises/nrf52-code/usb-app (main) :: cargo run --bin hello -- --allow-erase-all
   Compiling dk v0.0.0 (/Users/tallamjr/github/tallamjr/forks/rust-exercises/nrf52-code/boards/dk)
error: mutable reference of mutable static is discouraged
   --> /Users/tallamjr/github/tallamjr/forks/rust-exercises/nrf52-code/boards/dk/src/lib.rs:277:36
    |
277 |         ep0in: unsafe { Ep0In::new(&mut EP0IN_BUF) },
    |                                    ^^^^^^^^^^^^^^ mutable reference of mutable static
    |
    = note: for more information, see issue #114447 <https://github.com/rust-lang/rust/issues/114447>
    = note: reference of mutable static is a hard error from 2024 edition
    = note: mutable statics can be written to by multiple threads: aliasing violations or data races will cause undefined behavior
note: the lint level is defined here
   --> /Users/tallamjr/github/tallamjr/forks/rust-exercises/nrf52-code/boards/dk/src/lib.rs:6:9
    |
6   | #![deny(warnings)]
    |         ^^^^^^^^
    = note: `#[deny(static_mut_ref)]` implied by `#[deny(warnings)]`
help: mutable references are dangerous since if there's any other pointer or reference used for that static while the reference lives, that's UB; use `addr_of_mut!` instead to create a raw pointer
    |
277 |         ep0in: unsafe { Ep0In::new(addr_of_mut!(EP0IN_BUF)) },
    |                                    ~~~~~~~~~~~~~~~~~~~~~~~

error: could not compile `dk` (lib) due to 1 previous error

```

ryan-summers, 2024-03-11:
I just hit this while updating `usb-device` in https://github.com/rust-embedded-community/usb-device/pull/144 - the [quick fix](https://github.com/rust-embedded-community/usb-device/pull/144/commits/14de73e01c393db5361f38ac030b853d4bb24f84) in my case was to simply wrap the `static mut` buffer in an `UnsafeCell`. Hope that helps!

PR #85 (2024-03-14): Removed use of static mut.
Remove `static mut` values, our references to which were going to cause a warning in the next Rust release.

Uses the grounded crate, and a custom wrapper to make an !Sync type Sync.

Closes: #84
closed 2024-03-14 by jonathanpallant