ferrous-systems/rust-exercises #138: Cargo bloat issue

plaindocs, 2024-10-14:
When I run ` cargo bloat --bin hello`, following http://localhost:3001/nrf52-tooltips.html I get a pretty different output. Is this just OS differences (Fedora Linux) or a sign that something isn't quite right? 

```
❯ cargo bloat --bin hello
    Finished `dev` profile [optimized + debuginfo] target(s) in 0.06s
    Analyzing target/thumbv7em-none-eabihf/debug/hello

File  .text   Size          Crate Name
0.4%  29.9% 1.6KiB            std <&T as core::fmt::Display>::fmt
0.1%  10.6%   562B            std core::fmt::num::imp::<impl core::fmt::Display for u32>::fmt
0.1%   6.9%   368B          hello hello::__cortex_m_rt_main
0.1%   6.6%   350B            std core::fmt::write
0.1%   6.2%   328B            std compiler_builtins::mem::memcpy
0.0%   3.0%   160B      defmt_rtt _defmt_write
0.0%   2.6%   140B    panic_probe <&T as core::fmt::Display>::fmt
0.0%   2.6%   138B          defmt core::fmt::Write::write_char
0.0%   2.5%   134B      defmt_rtt defmt_rtt::channel::Channel::write_impl
0.0%   2.3%   124B    panic_probe rust_begin_unwind
0.0%   1.8%    98B            std core::result::unwrap_failed
0.0%   1.5%    80B      defmt_rtt _defmt_acquire
0.0%   1.5%    80B            std core::ops::function::FnMut::call_mut
0.0%   1.4%    76B      defmt_rtt _defmt_release
0.0%   1.4%    72B             dk nrf_hal_common::gpio::Pin<MODE>::into_push_pull_output
0.0%   1.3%    70B            std core::slice::index::slice_start_index_len_fail
0.0%   1.2%    66B            std core::fmt::Formatter::pad_integral::write_prefix
0.0%   1.2%    64B            std core::panicking::panic_explicit
0.0%   1.2%    62B    cortex_m_rt Reset
0.0%   1.0%    52B nrf_hal_common nrf_hal_common::ieee802154::Radio::wait_for_state_a
0.2%  12.6%   668B                And 38 smaller methods. Use -n N to show more.
1.3% 100.0% 5.2KiB                .text section size, the file size is 390.9KiB
```

robamu, 2025-10-07:
I received the following using `cargo bloat --bin hello --release `

```sh
File  .text   Size       Crate Name
0.4%  22.7% 1.5KiB         std <&T as core::fmt::Display>::fmt
0.2%  13.4%   884B       hello hello::__cortex_m_rt_main
0.2%  11.2%   744B         std compiler_builtins::mem::memcpy
0.1%   6.2%   412B         std core::fmt::Formatter::pad_integral
0.1%   6.0%   400B   defmt_rtt _defmt_write
0.1%   5.5%   366B         std core::fmt::write
0.1%   4.0%   268B   defmt_rtt _defmt_release
0.1%   3.5%   230B         std core::fmt::num::imp::<impl core::fmt::Display for u32>::fmt
0.1%   3.1%   206B         std <&T as core::fmt::Display>::fmt
0.0%   2.4%   160B   defmt_rtt _defmt_acquire
0.0%   2.1%   142B   radio_app _defmt_timestamp
0.0%   2.1%   140B         std core::fmt::Write::write_char
0.0%   2.0%   130B   defmt_rtt defmt_rtt::channel::Channel::blocking_write
0.0%   1.7%   114B       defmt defmt::export::fmt
0.0%   1.5%    98B         std core::result::unwrap_failed
0.0%   1.4%    92B   defmt_rtt defmt_rtt::channel::Channel::nonblocking_write
0.0%   1.1%    70B         std core::slice::index::slice_start_index_len_fail::do_panic::runtime
0.0%   1.0%    66B         std core::fmt::Formatter::pad_integral::write_prefix
0.0%   0.9%    62B cortex_m_rt Reset
0.0%   0.7%    48B          dk RTC0
0.1%   6.8%   452B             And 23 smaller methods. Use -n N to show more.
1.6% 100.0% 6.5KiB             .text section size, the file size is 399.5KiB
```

I used ` --release` as well. I think using `--release` is the easiest way to cut down on release size in the first place, so `bloat` is probably more interesting on the release version? Maybe the tooltip can be updated to use  `cargo bloat --bin hello --release ` as well. @jonathanpallant what do you think?

The size of these blocks probably changes frequently based on compiler and dependency version. The difference does not seem to extreme.

jonathanpallant, 2025-10-08:
Yes, use release. We can add a note about it changing version by version.